ZitaoTech/Hackberry-Pi_Zero #2: Release PCB design and BOM

iraqigeek, 2024-07-30:
Hi,
I found about the Hackberry via youtube, and wanted to make one, but I see you haven't released the design files. I saw you're selling fully assembled ones on your Tindie, but I wanted to make my own. Would you mind releasing the PCB design and BOM? Alternatively, just release the BOM and make the PCB available on your Tindie shop for people like me who want to make their own.
Thanks!

ZitaoTech, 2024-07-31:
Hello,

I will release the schematic but not the Gerber file, because some components are custom ordered like the display and the backlight control chip. Even if you have the file it will be difficult and will cost more than the price I set to make one yourself.

Zitao

securitylion, 2024-08-07:
Any chance you will sell just the custom PCBs so we can buy, print, and assemble ourselves?

LeonLeeXiang, 2024-08-12:
> Hello,
> 
> I will release the schematic but not the Gerber file, because some components are custom ordered like the display and the backlight control chip. Even if you have the file it will be difficult and will cost more than the price I set to make one yourself.
> 
> Zitao

你好，我也是在德国的留学生，正好想用BBkey做一个legion go 的手柄键盘试一试，请问键盘硬件是从国内订购的么？

ZitaoTech, 2024-08-12:
> 你好，我也是在德国的留学生，正好想用BBkey做一个legion go 的手柄键盘试一试，请问键盘硬件是从国内订购的么？

电路板我自己设计的 PCB是嘉立创生产的

ZitaoTech, 2024-08-12:
> Any chance you will sell just the custom PCBs so we can buy, print, and assemble ourselves?

The assembly is too difficult for a non-tech person to accomplish, actually

vicliu624, 2024-09-16:
could you sell the PCB separately on on your Tindie， because I can remove some of the parts from my old phone?

LuizGaronce, 2024-12-27:
Hello everything is fine?
Is it possible for you to make the Schematic, PCB and Gerber KiCAD files available, please? I can find replacements for the missing parts.

Thanks.   : )