ZitaoTech/Hackberry-Pi_Zero #8: Too slow :(

mattfaraday, 2024-12-26:
The device is awesome. But it's just too slow. You can't run browsers etc on it as it even warns you about the lack of RAM for the job. 
Have you tried any other single board computers the same form factor? like Radxa Zero 3 W

ZitaoTech, 2024-12-26:
That's right, the biggest issue is rpi02w has only 512MB ram, it would be much better if it has 1GB ram. Also, are you running on 64bit OS? I found 32bit OS should be faster on rpi2w and you can try dietpi OS, which is more lightweighted. And for browser I would recommend web browser in terminal like w3m. 
Radxa zero3w doesn't support the dpi interface. It's difficult to drive the screen

huuck, 2024-12-27:
Does dietPi come in a 32bit flavour? Could not find it anywhere on their website.

markuman, 2025-01-09:
I guess CM4 version would solve the issue.