ZitaoTech/Hackberry-Pi_Zero #9: Board upgrades?

0verambitious, 2025-01-02:
I understand this was designed specifically for the pi zero 2w
But have you put any thought into maybe a revision for using maybe a radxa or soemthing with more ram? 4gb of ram would be amazing!
Either way: amazing work you've done here. Love the device 🙏🏼🙏🏼

ZitaoTech, 2025-01-02:
Yes I designed another version for rpi4B. Just launched last week but haven't uploaded a video on YouTube

https://github.com/ZitaoTech/HackberryPi-4B

0verambitious, 2025-01-02:
Joined the waitlist on tindie! Very cool stuff. 💯💯

oneguynick, 2025-01-11:
Are any of the Zero 2w boards pin compatible with this? Has anyone tested?

nin187, 2025-02-04:
Can I swap RPi zero 2w into Orange Pi zero 2W(4GB ram)?

CodeHedge, 2025-02-04:
> Can I swap RPi zero 2w into Orange Pi zero 2W(4GB ram)?

No, they don't have the same test pad locations