ZitaoTech/Hackberry-Pi_Zero #15: After switch off my hackberry is not turning on. It's showing like blank screen

mrnextgen, 2025-02-27:


CodeHedge, 2025-02-27:
Have you checked your SD card? 

If you have a spare SD card I would follow the guide on the GitHub for installing an OS. Then booting into that as a sanity check.

mrnextgen, 2025-02-27:
Yes i have spare one . I write A new os in this SD card but still same .
Top side led not blinking. Keyboard light and keyboard not working.only
showing a blank screen

araskurd, 2025-03-07:
I have the same problem and it has not been solved for me yet :(

freitas666, 2025-03-19:
> I have the same problem and it has not been solved for me yet :(

Some update bro?

astearon, 2025-04-04:
I have the same situation. The SD has the stock RPi OS x64 version with "dtoverlay=vc4-kms-dpi-hyperpixel4sq" added to the config.
The white LED top indicator light does not turn on and the cpu is not even warm.