ZitaoTech/Hackberry-Pi_Zero #18: Black Screen on Kali Linux

freitas666, 2025-03-19:
I followed the procedure from https://github.com/ZitaoTech/Hackberry-Pi_Zero/tree/main/Screen. The Hackberry shows Kali Linux starting, but then the screen goes black and nothing shows. I have normal SSH access, but the screen stays black on the Hackberry itself. How can I resolve this?

freitas666, 2025-03-19:
Probably some bug in new kali linux 2025 (19/03/2025)...... i tested with old version and worked great!!!

sportyooz, 2025-04-09:
Im having the same issue changed OS not able to get screen to load connected thru SSH all the updated and drivers for the screen

venatiodecorus, 2025-04-25:
i got a black screen on the latest version so i downloaded 2024.4 and got it to boot to desktop, but wifi isn't working. what version did you use @freitas666 ?

kadirkurt, 2025-06-04:
## 💪 Fixing Display and Wi-Fi Issues on Kali Linux (Raspberry Pi Zero 2 W)

**Tested Image:**
`kali-linux-2025.1c-raspberry-pi-zero-2-w-armhf.img.xz`

This guide explains how to resolve two common issues when using Kali Linux on a Raspberry Pi Zero 2 W:

1. 💥 **Display goes black right before the desktop loads**
2. 📶 **Wi-Fi (wlan0) appears as "unmanaged" in NetworkManager**

---

### ✅ 1. Display Fix — Black Screen Before Desktop

If your DPI display (e.g. Hackberry Pi or HyperPixel 4) shows boot text but turns black when the desktop should appear, apply the following configuration.

Edit `/boot/config.txt` and replace the relevant section with:

```ini
# Enable custom DPI display (720x720)
dtoverlay=hyperpixel4

disable_overscan=1
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0

framebuffer_width=720
framebuffer_height=720

enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x5f026
dpi_timings=720 0 20 20 40 720 0 15 15 15 0 0 0 60 0 36720000 4

# IMPORTANT: Disable KMS to prevent black screen
# Comment out or remove the following line if present:
# dtoverlay=vc4-kms-v3d
```

> 🧐 Without disabling KMS, the system may boot to a black screen due to driver conflict with DPI displays.

---

### ✅ 2. Wi-Fi Fix — `wlan0` Unmanaged in GUI

If you see `wlan0` listed as **unmanaged** and can't scan or connect to Wi-Fi in the desktop environment:

#### 🧹 Step 1: Enable management of interfaces in NetworkManager

Edit the configuration:

```bash
sudo nano /etc/NetworkManager/NetworkManager.conf
```

Ensure it contains:

```ini
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=true
```

#### 🧹 Step 2: Clean `/etc/network/interfaces`

Edit:

```bash
sudo nano /etc/network/interfaces
```

And make sure it only contains:

```ini
auto lo
iface lo inet loopback
```

> Remove any lines referencing `wlan0`.

#### 🔄 Step 3: Restart NetworkManager

```bash
sudo systemctl restart NetworkManager
```

#### 🛠️ Step 4: Force `wlan0` to be managed (if still unmanaged)

If `wlan0` still shows as unmanaged, fix it manually:

```bash
nmcli device set wlan0 managed yes
```

---

### 📱 Now Connect to Wi-Fi

You can now connect using:

* `nmtui` (text-based interface)
* GUI (click the Wi-Fi icon and select your network)

---

### 🧚‍♂️ Tested With

* **Image:** `kali-linux-2025.1c-raspberry-pi-zero-2-w-armhf.img.xz`
* **Device:** Raspberry Pi Zero 2 W
* **Display:** HyperPixel 4-compatible DPI screen (720x720)
* **Kernel:** 6.x ARMHF (32-bit)
* **Interface:** XFCE